adriangibbons/php-fit-file-analysis #2: Forerunner10 missing altitude

wti3pvs, 2015-01-23:
Hi adriangibbons,
First, your fit file reader is great! Thanks for this.
However I have read a fit file created with Garmin Forerunner10 and the altitude information is missing. In my Garmin Connect account I can see the altitude. So I think that the altitude information is in the fit file.
I have also output the fild information - also no altitude exists. Could you check this issue please? Here is a downloadlink to my fit file: http://stephan.mein-sporttagebuch.de/userfiles/tempfiles/514B0111.FIT
Thank you very much in advance.
Kind regards,
Stephan

adriangibbons, 2015-01-24:
Hi Stephan, thanks for using php-FIT-File-Reader!

Have a read of Garmin's [Does my Forerunner 10/15 record or display elevation?](https://support.garmin.com/support/searchSupport/case.faces?caseId=%7Bbba24c70-868e-11e2-65d0-000000000000%7D) from the Support FAQ:

> The Forerunner 10 and 15 both record and display: time, distance, pace, and calories. These watches do not show or record elevation.
> 
> Both watches do record the GPS positions, (latitude/longitude) in the activity track. **When uploaded, Garmin Connect will use a digital elevation model to look up the elevation for each GPS position in the track and an elevation profile will be displayed on the Activity's page.**

The Forerunner 10 does not have a barometric altimeter sensor, so it is not possible for it to record true altitude data. It also does not record altitude data from the GPS, presumably because it can be quite inaccurate.

Therefore, it is not present in the file as shown by the *show_debug()* function, which shows the following 'record' data is present: timestamp, position_lat, position_long, distance, and speed. This matches the information from Garmin's FAQ. Though this has raised a question over whether "speed" for this file is actually [pace](https://support.garmin.com/support/searchSupport/case.faces?caseId=%7B1f542630-0d8a-11de-76c5-000000000000%7D) (i.e. the inverse of speed) and I'll need to look the formula I use to convert speed units.

To mimic the behaviour of Garmin Connect, I've create an example (link below) using [Google Elevation API](https://developers.google.com/maps/documentation/elevation/). This looks up the altitude for a set of latitude/longitude coordinates.

Here's the demo using the file you provided: [Forerunner 10 demo](http://www.adriangibbons.com/php-FIT-File-Reader-demo/forerunner_10.php) - nice run, btw! :+1: 

I'm interested to know how this compares to the map and altitude chart in Garmin Connect?

adriangibbons, 2015-01-24:
I have looked at your file some more and I am confident that the speed data recorded by the device is in meters per second, as opposed to 'minutes per kilometer' or anything else for **pace**. Therefore my formulae for converting units are OK as-is.

I have created a branch of the code, **'add-pace-option'**, which will add an option to convert speed data to pace if the user wishes. Pace will be in *minutes per km (min/km)* for **metric** units; and *minutes per mile (min/mi)* for **statute**.

I'll merge it into the master branch once it's ready. I don't think much coding will be required so hopefully I'll get this done today.

adriangibbons, 2015-01-24:
It's in the master branch :)

Have a look at the README.md for instruction/information on this.

wti3pvs, 2015-01-24:
Hi Adrian,

Thank you for your fast solution! Thanks also for the information about elevation using Forerunner10. The data shown by Google Elevation Api to my track seems correct.

Cheers,
Stephan

Professional Support should be like yours!

closed 2015-01-24 by wti3pvs