adriangibbons/php-fit-file-analysis #4: Composer package & submit to packagist

mjanda, 2015-07-02:
Hello,
would it be possible to add composer.json file and submit it to https://packagist.org for easier updates?

adriangibbons, 2015-07-03:
Hi @mjanda - I'd be happy to include a composer.json file but have no experience of creating them. If you do then perhaps you can create one and send a Pull Request?

My limited understanding of Composer is that it helps with dependency management. My project is almost a standalone file (php-FIT-File-Analysis.php), with the exception of the requirement to have the [Trader](http://php.net/manual/en/book.trader.php) PECL extension installed if you want to use some of the analysis that relates to cycling power meters.

At first glance, it would seem not straightforward to include PECL extensions - from the Composer project pages:
https://github.com/composer/composer/issues/383
https://github.com/composer/composer/pull/2898

@gfrogley has rewritten some of the project to do away with the dependency on the Trader PECL for the Simple Moving Algorithm (trader_sma). @gfrogley, if you submit a Pull Request then I'll test and merge this into the main project! Thanks for your contribution!

gfrogley, 2015-07-03:
> composer/composer#383
> composer/composer#2898

Hi Adrian
Sorry, I'm really, really new to all this GitHub thing and have absolutely no idea what a pull request is or how to go about it. Simply, I found your code and reworked it, testing it on a php5.4 platform to make sure it worked without the trader extension (Web hosts were not happy about doing this for a free or budget service). I can send you a copy of the code, which may be much easier I think!
Best
Graham

adriangibbons, 2015-07-05:
Hi @gfrogley - it's reasonable straightforward and what GitHub is all about. Check out some of the documentation for an overview of forking/branching and pull requests, etc. You'll need to download a git client - I use the GitHub for Windows and it works well for me!

If you don't think you'll get into GitHub then you can perhaps put the code on Dropbox and send a link to it. Or even save the file as .txt on your host and share that. I'm currently travelling but should get chance to look at it in the next few days. 

Cheers! :)

gfrogley, 2015-07-07:
Hi Adrian
I've had a look to try to submit my version of your file.... really do not
understand any of the jargon about this or how to go about 'master'  so
I've attached it for you as a php file.

The sma function to replace trader_sma is from line 1178 to 1198
I've added some HR Analysis function, like Training Peaks does, where a
Power Meter isn't present. This is on line 1068 to 1084.

I think it does all I need for the moment, I'll leave the updating to you.
Best
Graham

adriangibbons, 2015-07-08:
Thanks Graham! It looks like you emailed your response rather than through the GitHub the website. The attachment didn't come through so I can't look at the PHP file :'(

Please could you email it to adrian.github@gmail.com - Cheers!

adriangibbons, 2015-08-16:
Added a composer.json for phpunit dependency but not yet submitted to packgist.org - will look to do so in the next week or so...

adriangibbons, 2015-08-22:
composer.json updated and submitted!

https://packagist.org/packages/adriangibbons/php-fit-file-analysis

closed 2015-08-22 by adriangibbons