adriangibbons/php-fit-file-analysis #11: Undefined index: record while parsing FIT file

pwoo, 2015-08-13:
Hi Adrian,

Thanks for the recent fix. In addition to that issue, there is another problem I've been encountering with a specific FIT file. Unfortunately, I am still quite unfamiliar with the FIT spec and cannot pinpoint what the problem is during parsing. What I can tell you is that it gives the error:

```
Undefined index: record
```

And that it works in GoldenCheetah. I have uploaded the file to https://www.mediafire.com/?hgc99xr1oc5ic98

Thanks.

adriangibbons, 2015-08-13:
Hey, no worries!

I will take a look at your file later. Can you provide some context? E.g.
 - What device was used to record it?
 - What kind of activity is it?
 - What sensors were connected? HR, power, cadence/speed, foot pod etc. 

Cheers

adriangibbons, 2015-08-13:
I loaded your file up and outputted the debug information:
```php
<?php $pFFA->show_debug_info(); ?>
```
[image]

So the file doesn't appear to contain any valid messages as there isn't a mesg_num of 5120 in the FIT standard. However, I recognise the field definitions:
> 0	position_lat
> 1	position_long
> 2	altitude
> 3	heart_rate
> 4	cadence
> 5	distance
> 7	power
> 253	timestamp

So clearly something isn't working right as the mesg_num should be 20 for record data...

adriangibbons, 2015-08-13:
So the issue is that the **product architecture** for this file is Big Endian and the class as it stands assumes everything is Little Endian. The code lines 679 and 680 currently skip over the reserved and architecture bytes.
*Table 4-3. Definition Message Contents of the file 'D00001275 Flexible & Interoperable Data Transfer (FIT) Protocol Rev 1.7.pdf' in the FIT SDK.*

Sure enough if I replace:
```php
$this->file_pointer++;  // Reserved - IGNORED
$this->file_pointer++;  // Architecture - IGNORED
```
with
```php
$this->file_pointer++;  // Reserved - IGNORED
$architecture = ord(substr($this->file_contents, $this->file_pointer, 1));  // Architecture
var_dump($architecture);
die();
```

I get ```int 1``` (Big Endian) for your file and ```int 0``` (Little Endian) for any others that I try. Interested to know whether this file came from a Garmin (which?) or if it was exported from a website as I had not seen a Big Endian architecture file so far...

I have an idea of how to resolve this, but it will have to wait until the weekend!

adriangibbons, 2015-08-13:
You can see that I am currently only considering the Little Endian [PHP pack() format characters](http://php.net/manual/en/function.pack.php) C,c,S,v,l,V,f,d:
```php
private $types = array(
	0	=> 'Ctmp',	// enum
	1	=> 'ctmp',	// sint8
	2	=> 'Ctmp',	// uint8
	131	=> 'Stmp',	// sint16
	132	=> 'vtmp',	// uint16
	133	=> 'ltmp',	// sint32
	134	=> 'Vtmp',	// uint32
	7	=> 'Ctmp',	// string
	136	=> 'ftmp',	// float32
	137	=> 'dtmp',	// float64
	10	=> 'Ctmp',	// uint8z
	139	=> 'vtmp',	// uint16z
	140	=> 'Vtmp',	// uint32z
	13	=> 'Ctmp',	// byte
);
```

adriangibbons, 2015-08-13:
The record data field definitions from your file:
> position_lat (0)
> 133 = sint32 = l = signed long (always 32 bit, machine byte order)
> 
> position_long (1)
> 133 = sint32 = l
> 
> altitude (2)
> 132 = uint16 = v = unsigned short (always 16 bit, little endian byte order)
> 
> heart_rate (3)
> 2 = uint8 = C = unsigned char
> 
> cadence (4)
> 2 =uint8 = C
> 
> distance (5)
> 134 = uint32 = V = unsigned long (always 32 bit, little endian byte order)
> 
> power (7)
> 132 = uint16 = v
> 
> timestamp (253)
> 134 = uint32 = V

v has a Big Endian counterpart, n
V has a Big Endian counterpart, N
**l is always machine byte order, will have to have a think about this!**

pwoo, 2015-08-13:
The file is being exported via a mobile app still in development (not by me). I have been told that they are using PowerTap Power Meter for power, torque and speed, GSC10 for cadence and GPolar H7 for heart rate.

adriangibbons, 2015-08-14:
Thanks @pwoo - I suspected that the file had not been created with a Garmin due to it being Big Endian and the FIT Definition Messages it contains!

If you want to try the [php-FIT-File-Analysis.php from the endianness branch](https://github.com/adriangibbons/php-FIT-File-Analysis/blob/endianness/php-FIT-File-Analysis.php), I have added some changes today to make this file work with my class. The changes just need some more testing and thought before I merge the commits into the master branch.

[image]

adriangibbons, 2015-08-15:
Enhanced the ```fix_data()``` function to cater for all fields captured that were stored as unsigned ints that need to be converted to being signed as ```unpack()``` doesn't allow you to specify endianness for signed ints.

Merged into the master branch too (#13) as seems to be working fine. Let me know if all good and I wIll close the issue and delete the endianness branch.

closed 2015-08-16 by adriangibbons

pwoo, 2015-08-17:
Thanks. It's no longer crashing for this file.

However, it looks like file_id doesn't exist un data_mesgs. If I'm intepreting the debug output correctly (image attached), it should be there?

[image]

pwoo, 2015-08-17:
@adriangibbons, 

In addition to the above, I've attempted using a stripped down version of pFFA-Basic-Test.php to simply instantiate phpFITFileAnalysis (and nothing else):

FIT2015_8_8_ 5_41_44.fit
Time: 1.03 minutes, Memory: 40.75Mb

It's taking unexpectedly long to parse compared to typical files.

adriangibbons, 2015-08-18:
Hi @pwoo, I'll answer your two questions separately:

> However, it looks like file_id doesn't exist in data_mesgs (getting undefined index). If I'm interpreting the debug output correctly (image attached), it should be there?

It's a reasonable assumption that there *should* be corresponding **data** messages for a give **definition** message, otherwise why bother defining it? However, I don't see anything that would prevent someone from creating a valid FIT file that does not have data messages for a given definition message.

I'm confident that the **file_id** information is not contained in the file you have shared:
* I performed some (reasonably quick) advanced debugging of the file and could not see this information being dropped by the phpFITFileAnalysis class.
* I used the *FitToCSV.bat* program from within the Java folder of the FIT SDK to convert it to a [CSV file](https://dl.dropboxusercontent.com/u/21386/FIT2015_8_8_%205_41_44.csv) and there's no data messages, only a definition.
* I also just uploaded the file to Strava and noticed that the manufacture and product information (e.g. Garmin 500) is not included for the activity. I've deleted the activity as I don't want it included in my totals, but some very nice wattage there I see ;)

Cheers, Adrian.

adriangibbons, 2015-08-18:
> It's taking unexpectedly long to parse compared to typical files.

Yes, I am experiencing the same - but only with this particular file.

I think the FIT file is suspect here (but I may be wrong!). You mentioned that it was created by a mobile app being developed by someone else - how confident are they that the FIT files are being created optimally?

For example, the record definition messages are defined over and over again, often with just two fields e.g. one of heart rate, distance, power; and a timestamp. This is not very efficient and may be the cause of the performance degradation. Something more compact, such as the below may help, but dependent on the developer:

[image]

Also, I'd consider that without the ```file_id``` data message that it is not a valid FIT file as **Figure 9-1. Activity File** from *'FIT File Types Description, Rev 1.7'* suggests you need to have the definition and data messages. I note that other common/useful FIT fields are not included such as lap and session data.

Seems that the hold-up is in ```phpFITFileAnalyser::read_data_records()``` - but I don't have a good explanation without more investigation and I'm probably not going to do so unless I see a similar problem in FIT files created by other devices.

Hope the above helps explain what is going on and is perhaps useful to the developer of the mobile app!

Cheers, Adrian.

pwoo, 2015-08-18:
Thanks a lot Adrian, I've forwarded the analysis to the interested parties.

adriangibbons, 2015-08-19:
Great stuff - the manufacturer and garmin_product fields in the file_id message are enumerated types:

[image]

[image]

So the developer would need to request to be added to the FIT profile (if they're not already) so that they can populate this message with the value that relates to them as the manufacturer. It seems product is only relevant for Garmin devices but I may be wrong.

Cheers