adriangibbons/php-fit-file-analysis #12: fix_data() for position_lat and position_long looks terrible

adriangibbons, 2015-08-14:
[image]

RyanfCooper, 2015-08-14:
Hey I was just looking at this as well.  Each gap in the attached chart represents a fixed lat long value.  By the way I really appreciate the work you are doing and I'm testing it very heavily! If you need want and help testing or what now let me know. 

[image]

adriangibbons, 2015-08-15:
Thanks @RyanfCooper - is that a chart of speed vs distance? 260+km? nice!

Hopefully I'll make some progress this weekend - thanks for the offer of help testing, if you have spotted any bugs or have any ideas for enhancements then please share :)

I'm planning on setting up some PHPUnit tests, which should take care of a lot of the regression testing when I make changes in the future, but I'm sure there are still bugs lurking in my code already. What device(s) are you using? Cheers

RyanfCooper, 2015-08-15:
Mostly garmin files (500/10/20, 800/10, fenix3, 910xt) but have access to wahoo, power tap etc.  

That file was actually only 100km but I was calculating distance based on lat long so everytime there was a gap it was adding a lot of distance. Was trying to do power over distance with some special auto segmentation code I'm working on.  

Ryan

adriangibbons, 2015-08-15:
Squashed the bug. SIlly mistake:
```php
} else if ($missing_keys[$i] < $max_key) {
```
Should be:
```php
} else if ($missing_keys[$i] < $min_key) {
```

Resolved in this [interpolate_missing_data() typo](https://github.com/adriangibbons/php-FIT-File-Analysis/commit/fb25b2744931ef1fa0b5b6e7ae922e64ddc18eff) within #14 pull request.

All looks good now:
[image]

closed 2015-08-15 by adriangibbons

RyanfCooper, 2015-08-15:
Awesome.  Will keep testing.  Loving what you are doing!

RyanfCooper, 2015-08-15:
Would be cool to make significant digits an option.  For distance or lay long I'm going in and updating the rounding to more sig digits.  (For elevation correction later)

adriangibbons, 2015-08-16:
Hey @RyanfCooper, ```['record']['position_lat']``` and ```['record']['position_long']``` are rounded to 5 decimal places, which is accurate to somewhere between ~0.5m and ~1m. Given the accuracy of the GPS device itself, I'm not sure more significant digits for these makes sense - unless you're talking about the rounding of ```['record']['distance']```, which is currently to only 2dp?

[Wikipedia decimal degrees article](https://en.wikipedia.org/wiki/Decimal_degrees)

RyanfCooper, 2015-08-16:
Hey sorry for some reason I was thinking 5 was ~10 meters.  Thanks!