adriangibbons/php-fit-file-analysis #20: Undefined variable: record_header_byte

mjanda, 2015-08-24:
https://github.com/adriangibbons/php-fit-file-analysis/blob/master/src/phpFITFileAnalysis.php#L907-L909

```
    private function readDataRecords()
    {
        $record_header_byte;
        $message_type;
        $local_mesg_type;

```

I'll get error on those lines (with HHVM - not sure about normal PHP)

btw: thanks for all those changes, I just came from holiday and seen all the work you've done.

adriangibbons, 2015-08-24:
Hi @mjanda - thanks for raising. I don't use HHVM but had a quick Google. It would seem that [Hack is quite symbiotic with the PHP language, but there are some PHP features that are not supported by Hack and the type checker](http://docs.hhvm.com/manual/en/hack.unsupported.php). For example:
* "Undefined" variables
* References (e.g., function foo(&$x))

The first is easy to fix (by the way are there any more undefined variables? They are hard to find as are OK in PHP), but I am a little concerned about the second as I do pass by variable in quite a few places, e.g.:
https://github.com/adriangibbons/php-fit-file-analysis/blob/master/src/phpFITFileAnalysis.php#L1198

I guess you get errors for pass by reference too?

Cheers, Adrian.

mjanda, 2015-08-24:
I don't get any other error, but I use the class in a very limited way now. Basically I only load the file, read first record's timestamp and see if I have that activity in database. Rest of my processing is done in Go already. So I don't use interpolateMissingData().

As for more undefined variables, I think there are only two more in interpolateMissingData() - I've tried some regexp that should find them and it came with only those two (```\n\s*\$[^;\s(+]+;```).

adriangibbons, 2015-08-25:
I didn't think of the regex - thanks for that!

Think this can be closed following https://github.com/adriangibbons/php-fit-file-analysis/commit/c5bd03887d564d7c6c3620c8c0d38f4f37ab3655

closed 2015-08-25 by adriangibbons