adriangibbons/php-fit-file-analysis #28: develop getJSONObject() function

adriangibbons, 2015-09-08:
Would be useful to get data in a JSON format e.g. for use in dc.js

adriangibbons, 2015-09-11:
It's beginning!...

[image]

gfrogley, 2015-09-11:
That's really cool Adrian!
I've been looking at the dc.js site you mentioned too and had several thoughts. Can you get all the data out in json format? If so, then graphing the sets would be (ahem) straight forward. Seemingly all graphs can react to a filtered dataset. If one graph was changed then the others, quadrant analysis and map would follow suit. This would enable finer selection... A coach could select repeats of a hill and see the cadence, force and compare with subsequent ascents. HR and power pie charts would also be interesting to look at. I'll try to find some time this weekend to take a closer look at the code, but I need to get out on my bike for a change! I think this ability might even go beyond what training peaks premium can offer.
Cheers
Graham

adriangibbons, 2015-09-12:
So I haven't got it all working, laps with the map in particular are a bit off, but I thought I'd upload anyway: http://adriangibbons.com/php-fit-file-analysis/demo/quadrant-analysis.php

https://github.com/adriangibbons/php-fit-file-analysis/commit/d4cf743a6d49a68237989db31cb8e75f6a4a1f01 has included quadrant-analysis.php (and required files) in the demo folder. If you use a different FIT file with laps, you'll be able to filter the data based on laps. You can also click and drag on the two histograms!

Keen to here your thoughts and how we can make this better!

Enjoy the riding! I'm doing a 140km ride known as the Hell of the North in Western Australia because it's usually very windy :S

gfrogley, 2015-09-12:
> d4cf743

Check out cyclcoaching.ddns.net/dc.php

The map seems to go a bit screwy when segment data is selected. It looks
like the co-ords are rearranged during the import, but I might be wrong. If
you zoom in enough you can see blue criss-cross lines. It seems to get
muddled when latlngDim gets index again when a lap is selcted, but I cant
suggest a resolution, its way out of my experience!
Graham

gfrogley, 2015-09-12:
try *cyclecoaching* instead.... crap keyboard!

gfrogley, 2015-09-12:
been thinking, this question may help
http://stackoverflow.com/questions/30868801/crossfiltering-of-json-data-by-the-most-recent-time
but I have absolutely no idea on how to implement it.

adriangibbons, 2015-09-13:
Hey, no dramas. I am back from my 140km ride and a bit achey, but I managed to fix the issue. Basically, the issue lay with the fact that the lat/lng dimension (latlngDim) was being sorted by crossfilter.js - as it should:

> **# dimension.top(k)**
> Returns a new array containing the top k records, according to the natural order of this dimension. *The returned array is sorted by descending natural order*.

adriangibbons, 2015-09-16:
Need to add input to request which values should be returned in the JSON.

adriangibbons, 2015-09-16:
Closed by https://github.com/adriangibbons/php-fit-file-analysis/commit/36f398e7a4d91cbb520a3d34a154761304644d0b

closed 2015-09-16 by adriangibbons