adriangibbons/php-fit-file-analysis #36: this class can only handle normal headers

HenkV74, 2016-09-07:
When importing a garmin 820 fit file it errors with "this class can only handle normal headers":

if (($record_header_byte >> 7) & 1) {  // Check that it's a normal header
  throw new \Exception('phpFITFileAnalysis->readDataRecords(): this class can only handle normal headers!');
            }

Anyone has updated code?

Included a copy of the problem fit file.

[niels probleem 820 2016-09-07-05-49-08.zip](https://github.com/adriangibbons/php-fit-file-analysis/files/460199/niels.probleem.820.2016-09-07-05-49-08.zip)

Regards,
Henk

adriangibbons, 2016-09-08:
Hi @HenkV74 

I will take a look at this when I get chance. I had not catered for headers that are not normal because at the time I developed this class:

- The documentation was not clear on what compressed timestamp headers looked like.
- I hadn't come across a single FIT file that had any contained in them. Thanks for providing one and congrats on your Garmin 820! :)

I think the last version of the standard I reviewed as at around v16. I see that it is now up to (v20)[https://www.thisisant.com/resources/fit].

I will try take a look at the weekend but I can't promise anything. If you have any thoughts or find a solution for processing compressed timestamp headers then let me know as I would be keen to pull this into my project.

Thank you for raising this, Adrian.

HenkV74, 2016-09-08:
Hi Adrian, I invested already quite some time in trying to solve this matter but I think that you can confirm that fit-files are not exactly the most developer friendly protocols. I started using your code some weeks ago as the fit routines I wrote some years ago in the early days of the fit-files started to have some problems with newer devices and I thought that this code was more up to date. Had quite some issues in using it but was able to addapt. 

I'll continue to try to solve this matter but for me also must try to find some time and not the least because of the fact that I must completely reread and try to understand the fit protocol.

So if in the near time you're able to find a solution, many thanks beforehand :).

adriangibbons, 2016-09-08:
I've been looking at this issue for a few hours and after initially thinking that it was quite straightforward to cater for compressed timestamp headers, now I am not so sure...

I tried uploading the file to a couple of sites, just to make sure that the FIT file is OK:

- Garmin Connect: it loaded up fine. To be expected as it is their standard.
- Strava: data loaded in OK, but something went amiss with the lat/longs as the map was a straight line somewhere near Africa!

I am banging my head against the screen as I don't know why I cannot get it to work.

I thought something to do with the 64 bit types that have been introduced, but don't think that is the issue after adding them...

I thought something to do with ability to have "Developer Data", but I don't think that is causing the issue either. There is Developer Data messages in there, but my class seems to stepping over them appropriate by moving the file_pointer on...

So frustrating! Will have a look again another time...

Adrian.

HenkV74, 2016-09-09:
I looked again for some time and have came up with a bit and think to know where the problem is. You're much more used to your fit code so maybe we can work it out together.

So first get rid of the throw 'can only handle normal data'

 if (($record_header_byte >> 7) & 1) {  // Check that it's a normal header
//	throw etc
	$timeOffset = $record_header_byte & bindec("00011111"); // value 0 to 31
	$local_mesg_type = $record_header_byte & bindec("01100000") >> 6;  
	$message_type = DATA_MESSAGE;
	$compressedHeader = true;
}
else
{
$message_type = ($record_header_byte >> 6) & 1;  // 1: DEFINITION_MESSAGE; 0: DATA_MESSAGE
$local_mesg_type = $record_header_byte & bindec("00001111");  
$compressedHeader = false;
}

The problem which arises is that we have a new $local_mesg_type . It is normaly initialized in the Switch: DEFINITION_MESSAGE. However a compressed record does not have a definition message, only a different local message. I could execute this portion but this piece of code does also read the global message which is wrong in the case of a compressed record.

So the code must be adapted to define this new structure en read on. If we've got the correct definitions we read just like the switch; DATA_MESSAGE does. However we will not get a timestamp. This is where the $compressedHeader comes in place. If $compressedHeader it should take the last known timestamp in the fit file and just add the $timeOffset to it. Then we would be allright.

Hope this helps you a bit to get this thing running. I'll look further into it next week but for now my time is up :( Hope you'll be able to do something.

Regards,
Henk

adriangibbons, 2016-09-10:
Hi @HenkV74

So, the issue was not there are Compressed Timestamp Headers in your FIT file, but the `php-fit-file-analysis` did not handle Developer Data, added in version 1.9 of the FIT protocol in April 2016. I'll update the title of this issue for posterity.

So the `$file_pointer` was not being updated correctly when Developer Data definition and data messages were being encountered and things were going wrong as bytes were being interpreted and processed incorrectly.

Following #38 I think we are all good now. Please could you test and let me know if things work?

Thanks and enjoy! :)

HenkV74, 2016-09-10:
It's ok now! Thanks. So it wasn't afterall a compressed data problem but the filepointer went simply out of sync and left me te believe that it was a compressed header. Thanks for updating!

Maybe you could add also line 850: 
                30 => ['field_name' => 'left_right_balance',          'scale' => 4,    'offset' => 0,   'units' => ''],

Regards,
Henk

closed 2016-09-10 by HenkV74

adriangibbons, 2016-09-12:
Hey @HenkV74 

commit 83335292d6b1f3f37dc5bc00f6dd2c63c57a2c0a has included some additional Record message field definitions, including:

> 30 => ['field_name' => 'left_right_balance', 'scale' => 4, 'offset' => 0, 'units' => '']

I'm a bit reticent to create a new release just for this one commit, but I could do so if you're using a package manager for your project.

Cheers, Adrian.