adriangibbons/php-fit-file-analysis #39: Class does not support Compressed Timestamp Headers

adriangibbons, 2016-09-10:
If anyone receives:

> phpFITFileAnalysis->readDataRecords(): this class can only handle normal headers!

Please share an example FIT file and I'll use this for testing the handling of these timestamps. I'd be flying blind without a file to work so I'm not developing this until someone experiences the Exception above being thrown!

Adrian.

laufhannes, 2017-02-03:
@undertrained is currently trying to convert the current FIT parser of [Runalyze](https://github.com/Runalyze/Runalyze), which is using [mrihtar/Garmin-FIT](https://github.com/mrihtar/Garmin-FIT) (a perl module), to this one. Although there's a lot of other stuff to do in that branch, I already can see that message in our unit tests, thrown by [FR70-intervals.fit](https://github.com/Runalyze/Runalyze/blob/master/tests/testfiles/fit/FR70-intervals.fit). Feel free to use that for your testing and developing ;)

adriangibbons, 2017-02-11:
Thanks @laufhannes  I'll take a look when I get chance, but very busy right now. Very useful to have a FIT file with a compressed timestamp header 👍

rafaelnajera, 2017-03-22:
As far as I can tell there's nothing more to compressed TS than to calculate the record's timestamp with the most recent full TS and the record's offset using the formula given in the FIT documentation.  I just did a pull request with code that I think implements this feature, please have a look. 

BTW, thanks for the library, it's really useful.

adriangibbons, 2017-03-23:
@rafaelnajera Hey, thanks for the PR! I will get round to reviewing this properly soon. At first glance it looks good. I didn't think it was overly complex, just haven't had much free-time lately and up until recently no FIT files with compressed timestamps to test with!

adriangibbons, 2017-03-31:
Closing this issue following merging PR from Rafael. Will probably add a test or two for the new code and include a FIT file with a compressed timestamp. Thanks all

closed 2017-03-31 by adriangibbons