adriangibbons/php-fit-file-analysis #40: Matching up indexes when some values missing

robthompsonweb, 2016-12-20:
Great library.

How do you go about matching up the indexes when some are missing? 
See attatched - its for the 'lap' element from a swim.
Some of the 'total_distance' elements are 0 (resting period) but this means the indexes are missing from 'swim_stroke' making it pretty hard to match up the corresponding distance and stroke. Is this is a bug or is that the way the data comes through? I looked at the 'fix data' option but it only works for certain elements.

[image]

adriangibbons, 2016-12-28:
Hi, apologies for the delay getting back to you. Things have been quite hectic in the run-up to the holidays.

> Some of the 'total_distance' elements are 0 (resting period) but this means the indexes are missing from 'swim_stroke' making it pretty hard to match up the corresponding distance and stroke. Is [...]

It's the way the data comes through. Looking at the profile spreadsheet that comes as part of the download from the ANT developer area `swim_stroke` is an enumerated type where:
    0 = freestyle
    1 = backstroke
    2 = breaststroke
    3 = butterfly
    4 = drill
    5 = mixed
    6 = im

So there is no swim stroke that would correspond with a resting period.

`length_type` may be of some help. It is also an enumerated type:
    0 = idle
    1 = active

I must admit, I have used the class much for analysis of swim files. Perhaps you can share a link to a swim FIT file? I'd be interested in taking a look at it in more detail when I get the chance.

What device(s) do you use to record swims?

Cheers, Adrian.

laufhannes, 2017-02-03:
If I understand correctly, the problem of @robthompsonweb is not that `swim_stroke = 0` for some points, but that most arrays have seven elements but some have only four elements. It's probably okay if the missing ones are not set, simply because there's no data for them, but the indices are not consistent, as `swim_stroke[3]` probably relates to `total_distance[5]`.

For test files, there are at least five swim files in https://github.com/Runalyze/Runalyze/tree/master/tests/testfiles/fit, one from an outdoor swimming, two standard files for a 25m and 50m pool respectively and two files with additional data stored in developer fields.

robthompsonweb, 2017-02-10:
Sorry for late reply - @laufhannes yes that is exactly the issue.
It seems in order to match up the indexes you need to look at `first_length_index`  or `num_active_lengths` and jump if its 0.
Example FIT file for the swim above attatched - guess it would be nice if you set these indexes to null or something.
[example.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/766848/example.txt)

closed 2018-02-13 by robthompsonweb