adriangibbons/php-fit-file-analysis #42: Error: phpFITFileAnalysis->readHeader(): file_header['data_size'] does not seem correct!

robthompsonweb, 2017-04-21:
Hi again,

Run into the above error with a FIT file from a brand new Garmin Forerunner 395 watch. If you comment out the line that throws the error it will parse it but some information is missing.

Ie Record->heart_rate is not present (but it is present in the lap array). This information is definitely in the file as if you upload it to Garmin connect its shown.

This watch seems to have been added between FIT SDK versions 20.16 and 20.30 (latest) somewhere. Not sure if its stored in a different field in the fit file (as this watch uses a different way of capturing heart rate I belive)

[1627845922.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/946767/1627845922.txt)

robthompsonweb, 2017-04-21:
on further inspection seems the HR data comes through in the fit file in a new "hr" index. I will see if I can add this but any ideas about the header check - not quite sure what is going on their!

adriangibbons, 2017-04-25:
Hi Rob, I've had a quick look. Unfortunately the thisisant developer page is down so I can't download the latest version of the standard to see what is going on.

I did few ```var_dump()``` just before the exception in question is thrown.

```php
var_dump(strlen($this->file_contents));     // = 59548
var_dump($this->file_header['data_size']);  // = 32806
```

So we can see that the data size in the header is nowhere near the size of the file. I wonder if a load of new header gubbins and stuff is in the file now. Until I get my hands on the latest standard I can't progress things.

On the other issue, seems weird that they'd add a 'hr' index as well as 'heart_rate' to the record message definition!?

Thanks for reaching out! Adrian

adriangibbons, 2017-04-26:
Hey @robthompsonweb - I've downloaded the FIT standard just now and it doesn't look like anything has changed that should be causing this issue (albeit I've only looked for 10mins!):

[image]

It's a bit perplexing as the file size is ~59kb however ```$this->file_header['data_size']``` reports that the data size is ~32kb and should be roughly equal (i.e. 16 bytes different). I will look into how the data size is being read a bit more.

Some questions:

1. Do you get the same error when trying other FIT files?
2. Could you provide a couple more from your FR395?
3. Do you files upload OK to online services e.g. Strava, MapMyRide, Garmin, etc?

Cheers, Adrian.

adriangibbons, 2017-04-26:
Also, is it a Garmin FR 935? I can't find a 395 on the web so guess that was a typo? Nice looking piece of kit - I've still got an older Forerunner but don't really run enough to justify a new watch! :)

robthompsonweb, 2017-04-26:
Hey,

So its not actually my watch unfortunately - I was just getting the device type from file_id->product (fr935).
All other FIT files work fine - I assumed it was a problem with this one but have tried uploading it to Strava and Garmin connect and works fine.

Another example (from the same watch I think):
[1650639073.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/957759/1650639073.txt)

xjack913, 2017-12-04:
Hi, this file will show this error, too.
"phpFITFileAnalysis-&gt;readHeader(): file_header['data_size'] does not seem correct!"

Any solution? Thank you very much.

[7BTI2946.FIT.TXT](https://github.com/adriangibbons/php-fit-file-analysis/files/1526444/7BTI2946.FIT.TXT)

adriangibbons, 2017-12-05:
Hi @xjack913 

I'm working on a longer-term fix, but in the meanwhile you can just comment this code out so that it doesn't throw the exception. It will likely be related to heart rate data being in their own 'HR' messages, rather than where they've traditionally been: i.e. 'Record'->'Heart_rate', so you may not see these either.

Seems to be a pivot on Garmin's behalf as to their preference on capturing these fields. Unclear whether they will also split out other messages from under 'Record' into their own messages e.g. 'Power', etc.

Adrian.

adriangibbons, 2018-01-03:
Hi @xjack913 I've just made commits and up-versioned to v3.1.0 - does you file work OK now? Cheers, Adrian.

xjack913, 2018-01-04:
@adriangibbons Cool, thank you very much. I will be back after testing all those files.

closed 2018-02-13 by robthompsonweb