adriangibbons/php-fit-file-analysis #43: question: php fit 2 ways

bvbever, 2017-04-27:
It's great that your code analyse fit files.
is the opposite diretion possible: Is there a possibility to create fit files starting from some coordinates ?

adriangibbons, 2017-04-27:
Hi @bvbever

It is definitely possible to create (encode) a FIT file programmatically but this class does not do that. You'd use PHP's ```pack()``` rather than ```unpack()``` as this class does.

I'm pretty sure there are other projects here that enable you to encode a FIT file, though they may be in another language.

Do you need this functionality in PHP? What's the use case you want to do this for?

If there's nothing out there then I could potentially be tempted to collaborate with you. 

Cheers

bvbever, 2017-04-27:
Hi @adriangibbons 
There are indeed other projects that encodes a FIT file, but I need it in PHP. Our entire framework uses PHP. 
There is something like https://github.com/krwes/fit-php that uses somekind of older library.
cheers,

adriangibbons, 2017-05-01:
Hi @bvbever 
What's the framework? I went to your Github page but can't see your repos.
Cheers, Adrian.