adriangibbons/php-fit-file-analysis #45: FatalErrorException: Unsupported operand types

robthompsonweb, 2018-02-13:
Getting the following error when using the attached FIT file with HR:
(1/1) FatalErrorExceptionUnsupported operand types
--
in phpFITFileAnalysis.php (line 2713)

Using the latest master version - looks like $last_event_timestamp is an array of 2 elements rather than the expected integer but not sure how you match the correct version to use..

[error3.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/1720218/error3.txt)

adriangibbons, 2018-02-14:
Will take a quick look now but I think the wife is expecting some of my attention, given what day it is :)

Might be the weekend before I get to take a proper look...

Is the FIT file recorded with a different device? Cheers

robthompsonweb, 2018-02-14:
Haha no rush – was just flagging it.

 

Its from a Garmin fēnix 5X – the swim uploaded to garmin connect is https://connect.garmin.com/modern/activity/2497435759

adriangibbons, 2018-02-21:
The Garmin Connect page can't be viewed any longer.

Try inserting the following at [line 2703](https://github.com/adriangibbons/php-fit-file-analysis/blob/master/src/phpFITFileAnalysis.php#L2703)
```php
// Manually scale timestamps (i.e. divide by 1024)
$last_event_timestamp = $this->data_mesgs['hr']['event_timestamp'];
if (is_array($last_event_timestamp)) {
    $last_event_timestamp = $last_event_timestamp[0];
}
```
You might need to try ```$last_event_timestamp[1]``` if heart rate isn't working.

If this works then I'll commit the change to the project. Cheers, Adrian.

adriangibbons, 2018-02-28:
Hey @robthompsonweb - do thing now work as expected for the file(s) throwing the error?

Cheers, Adrian.

robthompsonweb, 2018-03-05:
Yes so I did try it with [0] and [1] (not sure what data this is) which removes the error but only seems to give HR for half the activity (but garmin connect displays it for the whole activity).

Looking at the field record->heart_rate the last timestamp is 1518258007 - which is only about 1/3 through the activity (record->timestamp[988] out out of 3424)
[Json export ](https://jsonblob.com/69daeb68-2074-11e8-893a-8f7d0a9abaae)of the data from the fit file in the original post.

The field hr->event_timestamp is a array with 2 elements which could have something to do with it (its only parsing the first event??) - looking at another file with HR hr->event_timestamp is a timestamp not array (attached
[single_event_hr_fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/1780873/single_event_hr_fit.txt)
).

Will try digging a bit to see if I can figure out why when got a second