adriangibbons/php-fit-file-analysis #46: from fit file to mysql db

danny-36, 2018-02-13:
Hi Adrian,

I love your php class and use it for a personal project to show my cycling data. 
Now I would like to put into mysql db all my fit files for a few years so I can create a custom reports and graph.

I would like to have your opinion on how to organize the mysql db tables.

Thanks.

Daniele

adriangibbons, 2018-02-14:
Hi @danny-36 - glad that you're getting some enjoyment out of it!

I actually created a project at home a while back that stored all my rides into a DB. I'll share the SQL here for creating the tables if that's any good for you!?

danny-36, 2018-02-14:
yes for sure, I'd like to see how you organized the data into db.

adriangibbons, 2018-02-21:
Hi, I actually deleted the project and can't find it :'(

You could try the following as a starting point:

```sql
SET SQL_MODE = "NO_AUTO_VALUE_ON_ZERO";
SET time_zone = "+00:00";
CREATE DATABASE IF NOT EXISTS `my_cycling_data` DEFAULT CHARACTER SET utf8 COLLATE utf8_unicode_ci;
USE `my_cycling_data`;

CREATE TABLE `rides` (
  `ride_id` int(11) UNSIGNED NOT NULL,
  `name` varchar(255) COLLATE utf8_unicode_ci NOT NULL,
  `distance` float NOT NULL COMMENT 'meters',
  `moving_time` int(11) NOT NULL COMMENT 'sec',
  `elapsed_time` int(11) NOT NULL COMMENT 'sec',
  `total_elevation_gain` float NOT NULL COMMENT 'meters',
  `start_date` varchar(20) COLLATE utf8_unicode_ci NOT NULL,
  `start_latitude` float NOT NULL,
  `start_longitude` float NOT NULL,
  `summary_polyline` text COLLATE utf8_unicode_ci NOT NULL COMMENT 'Google polyline',
  `kilojoules` smallint(5) UNSIGNED NOT NULL,
  `average_watts` smallint(5) UNSIGNED NOT NULL,
  `normalized_power` smallint(5) UNSIGNED NOT NULL,
  `variability_index` float NOT NULL,
  `intensity_factor` float NOT NULL,
  `training_stress_score` smallint(5) UNSIGNED NOT NULL
) ENGINE=InnoDB DEFAULT CHARSET=utf8 COLLATE=utf8_unicode_ci;

CREATE TABLE `ride_data` (
  `ride_id` int(10) UNSIGNED NOT NULL COMMENT 'Foreign key to rides.ride_id',
  `time` mediumint(8) UNSIGNED DEFAULT NULL COMMENT 'seconds',
  `distance` mediumint(8) UNSIGNED DEFAULT NULL COMMENT 'meters',
  `latitude` float DEFAULT NULL,
  `longitude` float DEFAULT NULL,
  `altitude` smallint(5) UNSIGNED DEFAULT NULL COMMENT 'meters of feet',
  `speed` float DEFAULT NULL COMMENT 'km/h or mph',
  `heartrate` tinyint(3) UNSIGNED DEFAULT NULL COMMENT 'bpm',
  `cadence` tinyint(3) UNSIGNED DEFAULT NULL COMMENT 'rpm',
  `power` smallint(5) UNSIGNED DEFAULT NULL COMMENT 'watts',
  `temperature` tinyint(3) UNSIGNED DEFAULT NULL COMMENT 'C or F'
) ENGINE=InnoDB DEFAULT CHARSET=utf8 COLLATE=utf8_unicode_ci;

ALTER TABLE `rides`
  ADD PRIMARY KEY (`ride_id`);

ALTER TABLE `ride_data`
  ADD KEY `ride_id` (`ride_id`);

ALTER TABLE `ride_data`
  ADD CONSTRAINT `ride_id_constraint` FOREIGN KEY (`ride_id`) REFERENCES `rides` (`ride_id`) ON DELETE CASCADE ON UPDATE CASCADE;
```
Hope this helps!

danny-36, 2018-02-26:
perfect, I try with this. Thank you.

danny-36, 2018-03-08:
The $pFFA->data_mesgs['file_id']['serial_number'] show a negative number (es: -513557248) this is a unique id for each file?

adriangibbons, 2018-03-09:
Looking at the ANT+ documentation, there isn't a whole lot said on ```file_id.serial_number```

> **4.8.1 File ID Messages**
> The purpose of the ‘file_id’ message is to uniquely identify the file in a global system. The fields in the data message may include file type, manufacturer, product, serial number, time created and file number depending on the FIT file type.

danny-36, 2018-03-09:
I have noticed that this number does not change if I read a different fit file.
I want to track all the files imported into the db. I have now created a custom identifier for all file.

I have tested a php script for import 20 files and there are no problems but how can I import all files of one year ride withoyt timeout php script problem?

closed 2018-04-23 by adriangibbons