adriangibbons/php-fit-file-analysis #47: v3.1 Unsupported operand types from the fit file from TomTom

xjack913, 2018-03-23:
I just updated the version to v3.1 and there is an error of "Unsupported operand types" 
in /home/ubuntu/workspace/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php at line 1446.

Could you please fix this or What should I do? Thank you.

[run-20180322T212609.fit.TXT](https://github.com/adriangibbons/php-fit-file-analysis/files/1840173/run-20180322T212609.fit.TXT)

adriangibbons, 2018-04-02:
Hi @xjack913  - see issue #45. Could you let me know if that works for you? 

I have started working on a fix. The commits are in the master branch but ahead of the latest release. I'll up-version once we know that the issue is addressed.

xjack913, 2018-04-03:
Hi, @adriangibbons I tried inserting the code issue #45 mentioned at line 2703, but it doesn't work.

I find out that the field_name "start_time" of each message_name "lap" of fit files from TomTom users is an array, not a timestamp. Therefore, when do add or sub operator to "this array" at line 1446 or 1493, it shows a Fatal Error Unsupported operand types.

Any suggestions? Thanks.

xjack913, 2018-04-24:
Thanks, these fit files from TomTom users work fine in v3.2. 

But a file show another issue, Division by zero in /home/ubuntu/workspace/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line <i>1962</i>

`$this->data_mesgs[$message][$field] = round(60 / 3.6 / $this->data_mesgs[$message][$field], 3);`

[610464.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/1940689/610464.fit.txt)

adriangibbons, 2018-04-24:
> But a file show another issue, Division by zero in /home/ubuntu/workspace/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line _1962_
> 610464.fit.txt [1]

Great, that's good to know.

Is this file throwing the new error uploaded to Garmin Connect, Strava,
etc? It's useful to see the information and charts shown on this kind of
site to try and ensure this class is producing the same info.

Cheers, Adrian.

$this-\u003edata_mesgs[$message][$field], 3);`\r\n\r\n[610464.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/1940689/610464.fit.txt)"}],"action":{"name":"View Issue","url":"https://github.com/adriangibbons/php-fit-file-analysis/issues/47#issuecomment-383787693"}}}

xjack913, 2018-07-05:
Hi, @adriangibbons ,

Recently, I got some fit files without session info, but other services like garmin connect and strava could handle these fit files as well. Could you please help me to take a look ? These fit files as below.

[2775379975.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2165344/2775379975.fit.txt)
[2780451464.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2165345/2780451464.fit.txt)
[2825367441.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2165346/2825367441.fit.txt)

Thank you very much.

adriangibbons, 2018-07-06:
Hi @xjack913 - I had a look at the first file and confirm that there isn't Session information being collected in there. I'm pretty sure that session is not a mandatory field/message - in fact I think pretty much everything in the ANT FIT protocol is optional. Perhaps the Lap field/message contains the information you are looking for?
```php
var_dump($position_lat = $pFFA->data_mesgs['lap']);

// Output
array(31) {
["timestamp"]=> int(1528942335)
["start_time"]=> int(1528941993)
["start_position_lat"]=> float(25.0486)
["start_position_long"]=> float(121.5494)
["end_position_lat"]=> float(25.04917)
["end_position_long"]=> float(121.54935)
["total_elapsed_time"]=> float(340.741)
["total_timer_time"]=> float(340.741)
["total_distance"]=> float(1)
["total_cycles"]=> int(508)
["time_in_hr_zone"]=> array(7) {
    [0]=> float(93.053)
    [1]=> float(248.07)
    [2]=> int(0)
    [3]=> int(0)
    [4]=> int(0)
    [5]=> int(0)
    [6]=> int(0) }
["message_index"]=> int(0)
["total_calories"]=> int(51)
["avg_speed"]=> float(10.566)
["max_speed"]=> float(10.613)
["total_ascent"]=> int(4)
["total_descent"]=> int(0)
["avg_vertical_oscillation"]=> float(86.7)
["avg_stance_time_percent"]=> float(34.15)
["avg_stance_time"]=> float(228.7)
["event"]=> int(9)
["event_type"]=> int(1)
["avg_heart_rate"]=> int(154)
["max_heart_rate"]=> int(160)
["avg_cadence"]=> int(89)
["max_cadence"]=> int(91)
["lap_trigger"]=> int(2)
["sport"]=> int(1)
["sub_sport"]=> int(0)
["avg_fractional_cadence"]=> float(0.578125)
["max_fractional_cadence"]=> float(0.5) }
```

xjack913, 2018-07-06:
Hi, @adriangibbons ,

I did try to collect what I need from the lap field, but all results show that all information including distance, total_timer_tiime, avg_speed.. etc are different with those shown at other services.

For example, the timer time filed of first file is 340.741 secs as you shown, but it is 1768 secs by Strava or Garmin connect which matched the user feedback. Thanks.

xjack913, 2018-07-12:
Hi, @adriangibbons ,

Sure. Thanks.

2775379975.fit
[](https://connect.garmin.com/modern/activity/2827370458)

2780451464.fit
[](https://connect.garmin.com/modern/activity/2844876841)

2825367441.fit
[](https://connect.garmin.com/modern/activity/2844877379)

xjack913, 2019-01-14:
Hi, @adriangibbons 

These are other examples without session info and detail info differ with the same file uploaded to garmin connect.

[2764794397.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2754138/2764794397.fit.txt)
https://connectus.garmin.cn/modern/activity/2827424537

[2780451464.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2754142/2780451464.fit.txt)
https://connectus.garmin.cn/modern/activity/2844876841

[2876706991.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2754147/2876706991.fit.txt)
https://connectus.garmin.cn/modern/activity/3298526012

Could you please help to fix these files as other services? Thank you very much.