adriangibbons/php-fit-file-analysis #48: Get Cadence field failed when Stryd Power IQ app installed

xjack913, 2018-04-16:
Hi, 

All values of cadence filed are 0, when Stryd Power IQ app installed.  Could you handle this problem? Thanks.

Stryd Power IQ app link:
https://apps.garmin.com/en-US/apps/660a581e-5301-460c-8f2f-034c8b6dc90f

The activity with Stryd Power IQ app:
[activityWithStydPowerIQApp.fit.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/1916045/activityWithStydPowerIQApp.fit.txt)

adriangibbons, 2018-04-18:
Hi @xjack913

My money is on it being related to Issue #42

Except related to cadence rather than heart rate i.e.

> It will likely be related to cadence data being in their own 'Cadence' messages, rather than where they've traditionally been: i.e. 'Record'->'Cadence'

Though I'll need to check this out more thoroughly!

adriangibbons, 2018-04-19:
OK, so doesn't look like it is similar to #42 

Look at the record messages:
```php
var_dump($pFFA->data_mesgs['record']['altitude']);
// array(1962) { [1522152947]=> int(6) ...

var_dump($pFFA->data_mesgs['record']['speed']);
// array(1962) { [1522152947]=> float(2.986) ...
```
I notice that they are indexed with the timestamps. However the cadence array is the right size (i.e. 1,962 data points), but they are all zeroes and indexed not by timestamps:
```php
var_dump($pFFA->data_mesgs['record']['cadence']);
// array(1962) { [0]=> int(0) [1]=> int(0) [2]=> int(0) ...
```
I'll need to look into this further to figure out what's going on...

Adrian.

adriangibbons, 2018-04-19:
It was due to developer data overwriting native fields. For some reason the dev data contained just a bunch of zeroes. Give v3.2.0 a whirl, all changes are committed in the master branch so you can just download the latest phpFITFileAnalysis.php if you're not using Composer.

Should work 🤞

xjack913, 2018-04-21:
@adriangibbons Thank you very much. All cadence values appear at v3.2.0!

cklong2kbomdic, 2018-07-13:
Hi
I have same issue. power start from index 0

adriangibbons, 2018-07-16:
Hi - assuming you are using the latest version (3.2.0), could you share a .FIT file? You may need to change the extension of the file to .txt so that Github will let you attach it! Thank you

cklong2kbomdic, 2018-07-16:
cadence: [1531521159] => 0
    [1531521160] => 0
    [1531521161] => 0
    [1531521162] => 0
    [1531521163] => 92
    [1531521164] => 0
    [1531521165] => 0
    [1531521166] => 24

power: [0] => 0
    [1] => 0
    [2] => 0
    [3] => 0
    [4] => 0
    [5] => 0
    [6] => 0
    [7] => 23
    [8] => 62
    [9] => 110

[4006_22621334870.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/2196659/4006_22621334870.txt)

thanks.

adriangibbons, 2018-09-12:
Hi @cklong2kbomdic - sorry I haven't had time to look at this. I guess you're saying that the power array needs to be indexed using the same as other fields e.g. cadence is using timestamp in your example above?

Have you been able to fix this yourself?

Cheers, Adrian.