adriangibbons/php-fit-file-analysis #49: Undefined offset: 25 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228

OlafvdSpek, 2018-09-11:
```
1228: $this->types = $this->endianness[$architecture];

Undefined offset: 25 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
Undefined offset: 139 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
Undefined offset: 97 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
Undefined offset: 113 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
Undefined offset: 119 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
Undefined offset: 109 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
Undefined offset: 108 in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1228
```

OlafvdSpek, 2018-09-11:
```
array_unique() expects parameter 1 to be array, integer given in vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1535
```

adriangibbons, 2018-09-12:
Hi @OlafvdSpek - obviously something is going quite horribly wrong! The $architecture should only be 0 or 1 for big or little-endianess.

Are you able to share a FIT file where this is happening? What type of activity and what device have you used to record it?

Cheers, Adrian.

OlafvdSpek, 2018-09-13:
Sure, what's your email address?

adriangibbons, 2018-09-16:
You can save a file as a .TXT and upload it here if that works? Cheers

OlafvdSpek, 2018-09-16:
I'd like the file to stay (semi) private..

adriangibbons, 2018-09-17:
adrian.github@gmail.com

OlafvdSpek, 2019-05-15:
Hi Adrian,

Any luck?

OlafvdSpek, 2019-09-30:
@adriangibbons Ping ;)

adriangibbons, 2019-10-05:
Hi @OlafvdSpek 

Are you still having issues? You'd shared two files with me via email:

**garmin.fit**
As discussed via email, neither of us is able to reproduce the error and everything seems OK.

**1.fit**
This doesn't seem like an activity is recorded in the file. This is confirmed if I upload to Strava and there isn't any data:

[image]

OlafvdSpek, 2019-10-08:
```
$ composer info
adriangibbons/php-fit-file-analysis v3.2.2 A PHP class for analysing FIT files created by Garmin GPS devices
```

```
$ md5sum *.fit
acc6b419f903165f9a6cc4dd8e8a60d2  1.fit
60df7628d0105886a16f63dfcfdc846c  2.fit
c4b2b452ad461e1415b6f337d3e4523b  issue-49.fit
```

```
$ php test-fit.php 1.fit 
PHP Warning:  array_unique() expects parameter 1 to be array, integer given in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1543

Warning: array_unique() expects parameter 1 to be array, integer given in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1543
```

```
$ php test-fit.php 2.fit  
PHP Notice:  Undefined offset: 14 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232

Notice: Undefined offset: 14 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232
PHP Notice:  Undefined offset: 73 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232

Notice: Undefined offset: 73 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232
PHP Notice:  Undefined offset: 14 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232

Notice: Undefined offset: 14 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232
PHP Notice:  Undefined offset: 14 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232

Notice: Undefined offset: 14 in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 1232
PHP Fatal error:  Uncaught Error: Unsupported operand types in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php:2720
Stack trace:
#0 /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php(1143): adriangibbons\phpFITFileAnalysis->processHrMessages()
#1 /home/forge/fit/test-fit.php(6): adriangibbons\phpFITFileAnalysis->__construct('2.fit')
#2 {main}
  thrown in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 2720

Fatal error: Uncaught Error: Unsupported operand types in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php:2720
Stack trace:
#0 /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php(1143): adriangibbons\phpFITFileAnalysis->processHrMessages()
#1 /home/forge/fit/test-fit.php(6): adriangibbons\phpFITFileAnalysis->__construct('2.fit')
#2 {main}
  thrown in /home/forge/fit/vendor/adriangibbons/php-fit-file-analysis/src/phpFITFileAnalysis.php on line 2720
```

OlafvdSpek, 2019-10-08:
> This doesn't seem like an activity is recorded in the file (which is only 494 bytes) - maybe something went wrong when recording this.

Hi Adrian,

Maybe, but shouldn't the lib be able to handle that without PHP warnings?
I've emailed you 2.fit, a file that causes other warnings / errors. Let me know if you can reproduce them.

adriangibbons, 2019-11-07:
Just made a quick commit that should fix the issue with 1.fit
Family, work, life, other projects have been preventing me from progressing the other issues.