adriangibbons/php-fit-file-analysis #52: Notice: Undefined offset: 84 in phpFITFileAnalysis.php on line 1228

LudoWeb, 2019-06-19:
Hi,

I have an issue with the parser.

`( ! ) Notice: Undefined offset: 84 in \vendor\php-fit-file-analysis\phpFITFileAnalysis.php on line 1228`

Do you have any idea about this issue ?
Many thanks

adriangibbons, 2019-06-20:
The offending line 1228 is:
```$this->types = $this->endianness[$architecture];```

It would seem that ```$architecture``` has a value of 84, which won't work as it should only be 0 or 1 depending on whether little or big-endianness should be observed.

What device was the FIT file created with? E.g. a Garmin Forerunner XX or whatever.

Do you only have the issue with one FIT file or many?

I can have a deeper look if you share a FIT file with the issue.

LudoWeb, 2019-06-20:
It's a Suunto device. I have the issue with one FIT only.

See the file in attachment.

Many thanks :)
[31333487945.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/3309298/31333487945.txt)

LudoWeb, 2019-06-25:
Hey @adriangibbons,

Were you be able to check the issue ?

adriangibbons, 2019-06-26:
Hi @LudoWeb I have looked into this issue for the past 3 hours and not got very far.

I don't understand why the architecture is coming up as something other than 0 or 1.

Only thing I can think of is that this Forerunner combined with the HRM belt is using compressed HR fields, which I don't think the class currently supports. I read the FIT documentation and unfortunately, it is not obvious to me how to implement this (assuming that it is the issue in the first place)!