adriangibbons/php-fit-file-analysis #53: Precision rounding of fields

co-wurker, 2019-06-22:
Forgive me if this is overly simplistic, however, I'm new to Github and not familiar with the way things typically work. This is more of a feature request...

The rounding of units is restrictive in some cases. For example, distance fields are rounded to 2 decimal places. In many cases, due to the restriction, the maximum data resolution is greater than the maximum GPS error at 95% confidence level. I would kindly suggest considering 3 place precision on distance fields, or perhaps a config option to allow the developer a little more flexibility.

I can edit my files to do what I need, but of course, I would prefer not to create my own custom branch of the code. 

Thank you for your work on this useful resource, Adrian. 

Cheers.

adriangibbons, 2019-06-24:
Hi @co-wurker and welcome to Github!

I'll look into it. Yes I agree that if you're using statute or metric units then distances are rounded to miles/kilometers to 2dp, which on the face of it doesn't seem very granular.

That said, for people's sporting activities it seems about the right level of detail and indeed I think the likes of Strava etc don't go below this when analysing activities.

I'll need to put some thought in to make sure I don't break things for people already using the class i.e. their apps may assume only 2dps.

In the meanwhile, you could use raw units - this would give you distance measured in metres to 2dp (however LatLngs will be in semicircles), or you can calculate distances in your app from the LatLngs if they are available.

If you code a good way of handling this in your own branch, you can always then create a Pull Request (PR) whereby I would review proposed changes and accept/comment on them before they are included in the source repo (i.e. mine). That's the beauty of Github, you can contribute and make projects better!

:D

co-wurker, 2019-06-26:
Thanks for the prompt response, much appreciated man.

So, I just made a quick change to your file for my own purposes at the moment. My only complaint about the 2dp is when it comes to doing any sort of calculations, say if/when resampling a path, there's some likelihood of any calculation being up to 10 meters off due to rounding. GPS error is typically within a radius of ~ 8 meters, so rounding errors plus inherent GPS error may lead to some undesirable fuzziness in the data.

For display purposes, 2dp is perfectly fine, but for the maths, I would feel a little better working from 3dp, then formatting my display data however I choose to. I completely understand your hesitation to make this change, however.

Separately, do you have any channel of communication to chat about topics of interest around FIT files?

adriangibbons, 2019-06-26:
No worries, makes sense! I'll work on an option but may be a little while before I go around to this.

Here's a couple of links to ANT/FIT resources:
https://www.thisisant.com/developer/ant/ant-fs-and-fit1/
https://www.thisisant.com/forum/

For resampling, I have used (david-r-edgar/RDP-PHP)[https://github.com/david-r-edgar/RDP-PHP] and had good results!