adriangibbons/php-fit-file-analysis #54: Issue about

LudoWeb, 2019-07-03:
Hello Adrian :)

I have an issue with the parser. Unfortunalty, `data_mesgs['record']` is null and i don't have any data.

This line seems to be the problem :
`$this->file_header['data_size'] = $this->file_header['crc'] - $header_size + 2;`

See attachment : [my-session.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/3355141/my-session.txt)

It's a session recorded with Garmin Forerunner 920XT.

Many thanks,

adriangibbons, 2019-07-04:
If you insert the following after that line, what do you get?
```php
var_dump($this->file_header['crc']);
var_dump($header_size);
die();
```

LudoWeb, 2019-07-04:
`$this->file_header['crc']: int 635
$header_size: int 14`

adriangibbons, 2019-07-05:
Hmmm, seems like a small file
$this->file_header['data_size'] = 635 - 14 + 2 = 623
I will try find some time at the weekend to investigate

LudoWeb, 2019-07-15:
Hey @adriangibbons,

Any idea about this issue and how to fix it ?

Thanks!

adriangibbons, 2019-07-19:
Are you able to share the file? You will probably have to change the file extension to .txt

Presumably the file works fine with Garmin Connect / Strava / etc?

LudoWeb, 2019-07-19:
Yes, the file seems to work fine with Garmin Connect and Strava.

Here's the share file: [error-file.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/3411266/error-file.txt)

Thanks,

LudoWeb, 2019-08-06:
Hello @adriangibbons,

Were you able to check the issue ?

adriangibbons, 2019-08-08:
Hi, I had a quick look but didn't get anywhere. Then I've been on holiday a couple of weeks. Back in about a week or so and then will have more time. Have been so busy!