adriangibbons/php-fit-file-analysis #56: getJSON function for laps or other fit sections

lnovali, 2019-09-06:
Is it possible to have new getJSON functions for other FIT sections? This function extracts only data from "record" section, but I need also an extractor for "lap" section.
Thank you so much for this helpful lib!

adriangibbons, 2019-09-09:
Hi @lnovali - off the top of my head it should be fairly straightforward to enable this. It might have to be through a call like getJSON("lap") or getJSON("all") to prevent breaking the API for others.

I'm very busy these days with other projects, but I'll try get round to this soon.

adriangibbons, 2019-10-05:
@lnovali have a look at the ```getJSONLap()``` method I just added in f6427a0ad9997b4af053019d46ea3feb0326318c

I've this meets your purpose then I'll roll into a new release.

Initially I tried to create something more generic, but got tied-up in knots.