adriangibbons/php-fit-file-analysis #58: Suggestion

vallsjm, 2019-12-02:
I will suggest a litte change:

I have a REST api, that api have a method in order to import a FIT file. I am using this class after import, but, is imposible load RAW content. I need first write the content in a file and after that load with phpFITFileAnalysis.

Have sense a function like that?
`loadFromFile($file_path, $options = null)`
`loadFromData($data, $options = null)`

adriangibbons, 2019-12-08:
Hi @vallsjm 

Sounds good, though I thought that @roznet had added such functionality in #55 

See 4899406af62979c86b9a8d9eb0275017047d6ee2 where ```__construct($file_path, $options)``` became ```__construct($file_path_or_data, $options)```

Though I recall that @roznet was loading data from a database rather than an API. Not sure of the differences in how this manifests itself though, if any.

Do you have any thoughts on what the implementation of a ```loadFromData()``` method would look like?

Cheers, Adrian.

roznet, 2019-12-08:
Yes, I added that functionality exactly for this type of use case. But because I don't think (or know how) it's possible to have constructor in PHP with two type of inputs, I use the same option with an option to indicate the input is data as below:

```php
$fit = new adriangibbons\phpFITFileAnalysis( $data, array( 'input_is_data' => true ) );
```
where `$data` either is result of querying a database or from a REST api, in my case the file is with header: `Content-Type: application/octet-stream`

Hope it works for you as well

vallsjm, 2019-12-10:
Hi guys,

I think this is a great solution. Enough for me. It's more elegant than loadFromData.

Thank you very much!!

closed 2019-12-10 by vallsjm