adriangibbons/php-fit-file-analysis #59: incorrect speed values

alexfraundorf-com, 2020-01-09:
Hi Adrian,
I am having a problem with speed values that I was hoping you could help me hunt down.  I am getting a larger value for average speed than I am for max speed when I var_dump data_mesgs['lap'].
Below is what I am getting for those two sets.  I have the fit file available here: https://www.dropbox.com/s/ep31ebzc7s1izu3/file.fit?dl=1
Thank you,
Alex

["avg_speed"]=>
  array(8) {
    [0]=>    float(5.457)
    [1]=>    float(5.002)
    [2]=>    float(6.198)
    [3]=>    float(4.108)
    [4]=>    float(6.217)
    [5]=>    float(4.052)
    [6]=>    float(5.842)
    [7]=>    float(3.766)
  }
  ["max_speed"]=>
  array(8) {
    [0]=>    float(4.868)
    [1]=>    float(4.88)
    [2]=>    float(4.908)
    [3]=>    float(3.502)
    [4]=>    float(3.987)
    [5]=>    float(3.801)
    [6]=>    float(3.841)
    [7]=>    float(3.573)
  }

alexfraundorf-com, 2020-01-09:
This screenshot shows what should be the correct values according to https://www.fitfileviewer.com/
[image]

adriangibbons, 2020-01-09:
Hi Alex, I’m on holiday at the moment and can’t do much investigation. If you upload your file to a service like Garmin Connect or Strava (etc), do you get the issue present itself there?

alexfraundorf-com, 2020-01-09:
Hi Adrian,
No, the issue seems to only be with in this library.  It shows correctly on Garmin Connect and fitfileviewer.com.  It is also not just this particular fit file, but a problem I am having consistently.  I am wondering if the settings I am using could be the problem or if the units are not being loaded properly.
These are the options I am using:
$options = [
    'fix_data'                => ['all'],
    'data_every_second'       => false,
    'units'                   => 'metric',
    'pace'                    => true,
    'garmin_timestamps'       => true,
    'overwrite_with_dev_data' => false,
];
I hope you enjoy your holiday.
Thanks!
Alex

alexfraundorf-com, 2020-01-09:
I set the $options parameter to null and now I get the correct speed values in km/hour:
  ["avg_speed"]=>
  array(8) {
    [0]=>    float(10.994)
    [1]=>    float(11.995)
    [2]=>    float(9.68)
    [3]=>    float(14.605)
    [4]=>    float(9.652)
    [5]=>    float(14.807)
    [6]=>    float(10.271)
    [7]=>    float(15.934)
  }
  ["max_speed"]=>
  array(8) {
    [0]=>    float(12.326)
    [1]=>    float(12.294)
    [2]=>    float(12.226)
    [3]=>    float(17.132)
    [4]=>    float(15.048)
    [5]=>    float(15.786)
    [6]=>    float(15.62)
    [7]=>    float(16.794)
  }
So it seems that something in the $options array is causing my problem.  
Did anything jump out at you?
Again thank you for everything.
Alex

alexfraundorf-com, 2020-01-09:
Adrian,
It turns out that it was the option of 'pace' set to true that was causing the speeds to get messed up.
I'm not sure why, but this works as a fix, so there is no rush on this issue on my account.
Enjoy your holiday and thanks again!
Alex
The following options get me the correct data:
$options = [
    'fix_data'                => ['all'],
    'data_every_second'       => false,
    'units'                   => 'metric',
    //'pace'                    => true,
    'garmin_timestamps'       => true,
    'overwrite_with_dev_data' => false,
];

alexfraundorf-com, 2020-01-09:
Or is this not a bug, but my misunderstanding of how the pace option works?
If the pace option is set to true, is it the pace value in minutes being returned in the average_speed and max_speed arrays?

adriangibbons, 2020-01-10:
Ah yes, the max and averages will be affected as pace overwrites the speed data. From the README:
> **Pace**
If required by the user, pace can be provided instead of speed. Depending on the units requested, pace will either be in minutes per kilometre (min/km) for metric units; or minutes per mile (min/mi) for statute.

Typically you'd use speed (unchanged) for a bike ride and pace for a run. Do you use both together?
With retrospect, there is a case for leave speed unchanged and adding pace as a separate field rather than overwriting speed. It would be a "major" change to the class as the expected behaviour for people already using pace...

alexfraundorf-com, 2020-01-10:
My apologies Adrian.  I completely missed that in the README.
Having access to both speed and pace would be useful, but I agree that it could cause problems for current users.
Currently I am getting speed from your library and calculating pace, which is a simple calculation.
Thank you again for your library and sorry for wasting your time on this.

closed 2020-01-10 by alexfraundorf-com