adriangibbons/php-fit-file-analysis #61: data_mesgs['session'] is null

conejog, 2020-03-24:
Hi, im testing your parser (great job) but i have a problem.

I have an issue with the parser. Unfortunatly, data_mesgs['session'] is null and i don't have any data.
 data_mesgs['record'] is ok, full of data. the fit file is ok, is a bike workout (2 hours duration).
recorded using forerunner 935. i tested the fit in fitfileviewer.com and is ok.
do you know where is the problem?

thanks so much.

adriangibbons, 2020-03-25:
Hi, my understanding of the FIT protocol is that there are not many mandatory fields. I don't think a valid FIT file is required to have Session messages within it.

Is it just for one file (for your Garmin 935) that you get this or all? As a workaround, you can probably derive the information in Session from the Record data.

If you're able to share a FIT file here then I can take a look. I will run it through the official [FIT SDK](https://www.thisisant.com/developer/resources/downloads) Java program "FitToCSV" to generate a CSV file

conejog, 2020-03-25:
hi , here is the file
seems ok, i can see all the data in training peaks, garmin connect and fitfileviewver.com.
i don't know where is the problem....
[77212839306.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/4382645/77212839306.txt)

conejog, 2020-03-25:
i have a new one (from other device) with a similar problem.
says Uncaught Exception: phpFITFileAnalysis->readHeader(): not a valid header size!
but the file is ok
[78091779901.txt](https://github.com/adriangibbons/php-fit-file-analysis/files/4383449/78091779901.txt)

adriangibbons, 2020-04-01:
Just ran the x06.fit file through the SDK parser and there is ```session``` messages in there so I am investigating...

adriangibbons, 2020-04-01:
First look and I'm a bit stumped. Looking at the protocol to see if anything has changed and I note:

> An activity file shall contain file_id, activity, **session**, and lap messages. The file may also contain record, event, length and/or hrv messages.

So there definitely should be ```session``` information in the file (confirmed by the FitCSVTool.jar). I just don't see any messages with a global message number of 18, which relate to session messages.

conejog, 2020-04-16:
Hi, how are you?
do you have any update?
thanks. greetings,