adriangibbons/php-fit-file-analysis #63: Multiple session

Xring-git, 2020-07-28:
Hi,

for the first time Garmin (yesterday and today) has sent me a FIT with multiple sessions, that generate an error.

How to approach this?

Thanks a lot,
Franco

adriangibbons, 2020-07-29:
Hi @Xring-git I haven't seen a file with multiple sessions before either. Is a session related to multisport activities e.g. a triathlon contain swim-transition-bike-transition-run?

Are you able to share it so that I can take a look?

Xring-git, 2020-07-29:
Hi Adrian,

see in attach.

[garmin_activity.zip](https://github.com/adriangibbons/php-fit-file-analysis/files/4995030/garmin_activity.zip)

Eric-11, 2022-10-10:
Just curious if you have any insights into how to address multisport FIT files.

adriangibbons, 2022-10-10:
Happily, the Garmin documentation seems to have come one quite a bit over the past few years. Check out https://developer.garmin.com/fit/file-types/activity/ and see **Example Message Ordering for Summary Last File Recording of a Multi-Sport Activity** in particular.

I'd also use the [FitCSVTool](https://developer.garmin.com/fit/fitcsvtool/)

Eric-11, 2022-10-12:
I've updated the enum data and found that fixed the problem with the attached file.