adriangibbons/php-fit-file-analysis #64: Suunto Ambit Support

SBrownAU, 2020-09-24:
I'm trying to process a file that I can view by uploading to https://www.fitfileviewer.com/ If I try to process using this library I get `Undefined offset` error @ `$this->types = $this->endianness[$architecture];` 

`$architecture` is showing as `14`. Obviously, the code expects zero or one. Any idea? Fit file attached.

[68270818303.zip](https://github.com/adriangibbons/php-fit-file-analysis/files/5279200/68270818303.zip)

adriangibbons, 2020-09-28:
Hi, can you try commenting-out line 1343
https://github.com/adriangibbons/php-fit-file-analysis/blob/master/src/phpFITFileAnalysis.php#L1343

Appears to be a swimming fil with a Garmin 920XT. Were there any sensors attached e.g. HR strap?

SBrownAU, 2020-09-28:
Hmm yep looks like's a Garmin. Commenting out 1343 worked in allowing me to parse the file (and other similar ones I've had issue with). It's not my file so I'll try and find out whether the user had any sensors attached.

adriangibbons, 2020-09-28:
No worries. As you can read from the comment on the line before it was a bit of a fudge to ensure all data is processed. Otherwise "buffered messages" were not being parsed. I haven't kept up with changes to the FIT protocol standard for some time. Out of interest, what are you using the PHP class for? Is there a URL of a site I could visit? To be honest, I thought this project had gone stale and is quite old / on the verge of being made read-only as I don't really maintain it any longer.

SBrownAU, 2020-09-28:
I'm using it to process FIT files from the Garmin API. The project I'm involved with does our own processing and analysis on the files (some of which Garmin doesn't provide hence why we're processing ourselves). Public/marketing site is https://athletica.ai/ private site in beta at the moment.

adriangibbons, 2020-09-28:
Nice! Would you mind dropping me a line at adrian.github@gmail.com ? I'm interested in knowing about the Garmin API. I plug into the Strava API for my main project. After access to the Strava API, I didn't really need to process FIT files myself, hence the drop-off in maintaining this project.