adriangibbons/php-fit-file-analysis #65: Garmin Product Information is incorrect

riccode, 2020-10-08:
In the FIT files there are two fields "product" and "garmin_product" which are not interchangeable. Therefore, when pulling out the device information for Garmin devices there is an error. The code does not distinguish between the "product" fields and the "garmin_product" fields in the FIT file. This leads to errors. I reached out to Garmin directly and they noted that "product" has something to do with Garmin Connect and does not reference the product list from the FIT SDK. 

Therefore,  $this->data_mesgs['device_info']['product'] should NOT contain values pulled from the "product" field.

adriangibbons, 2020-10-10:
Hi @riccode 

The issue is thanks to it not be clear whether the field contains ```product```, ```garmin_product``` or ```favero_product``` as they all have the same field number in a ```device_info``` message.

From the Profile.xlsx file in the [FIT SDK](https://www.thisisant.com/resources/fit-sdk)

[image]

Same goes for ```file_id``` messages. Not sure how this can be overcome - i.e. how do we know which of the three Field Names is present in a ```device_info``` message? I don't see any clue in the documentation.

Cheers, Adrian.

riccode, 2020-10-14:
Adrian,

I reached out to Garmin, who in turn reached out to the FIT team and have made them aware of this. We will wait and see if they get back to me. So far the best solution I've found is using the software_version field. It seems that Garmin devices have a software_version below 1000, while software versions relating to 'product' had a software_version > 10,000. 

riccode