adriangibbons/php-fit-file-analysis #69: Demo not working

rucksman, 2022-06-01:
I just discoivered your project. Unfortunateyl I can't get it to work.

What I did:
- Cloned yor repo to my webserver
- Copied one of my fit-files (Garmin Edge 530) to the webserver
- Renamed this fit-file to mountain-biking.fit

When I go to demo/mountain-biking.php, there is nothing to see in "Flot Charts". In Google Map there is only an error message concerning Google Geocoding API and Google Time Zone API.

When I go to demo/mountain-biking-leaflet.php, there is nothing to see in Flot Charts and Leaflet Map.

What am I missing here?

On top of the page the device is not recognized ("Garmin unknown"), the other information like Duration and Distance seems to be correct).

rucksman, 2022-06-01:
OK, in demo/mountain-biking-leaflet.php the problem is due to an error around line 215. In developer tools in the browser I get this error message:
```
SyntaxError: expected expression, got '<'
```
And after inspecting the source code of the page, I see:
```
var heart_rate = {
      'color': 'rgba(255, 0, 0, 0.8)',
      'data': [
<br />
<b>Warning</b>:  Invalid argument supplied for foreach() in <b>/path/to/demo/mountain-biking-leaflet.php</b> on line <b>220</b><br />
      ]
    };
        	
    $.plot('#speed', [speed], speed_options);
    $.plot('#heart_rate', [heart_rate], heart_rate_options);

```

After commenting out the php part in the variable heart_rate, everything works fine. I have to say that during the trip which reflects the fit file, I dod not use a heart rate monitor. But why does this make such problems?

adriangibbons, 2022-06-04:
Hi! Yes, the project is really quite old and no longer maintained.

The API issue is likely because the Google APIs now require a key as part of the request or Google has changed something else that broke things.

The demo file has heart rate, which is why the demo tries to draw a chart with data regardless of whether there is HR data or not. You could probably use an ‘’’empty()’’’ on the relevant array to better handle drawing/not drawing of charts depending what’s in the FIT file. It’s not intended to be a complete work solution, just a quick demo 👍

rucksman, 2022-06-05:
Thanks for answering! Yes, the map problem is surely about Google requiring a valid API key, so this not really an issue. I will try to tweak the demo files so that it does only draw graphs for data which is actually present. Anyway, thanks for this project!

closed 2022-06-05 by rucksman