adriangibbons/php-fit-file-analysis #70: Google Geocoding API: Error

sergiocfv, 2023-04-16:
Why this error appear?

adriangibbons, 2023-04-17:
I assume you are looking at one of the examples? They are very old.
If you want to use the Google Geocoding API, you probably need to replace the `API_KEY` with one you have created for yourself.
https://developers.google.com/maps/documentation/geocoding/overview

sergiocfv, 2023-04-17:
Thanks.
Yes I've look the example. Is possible use it without Google API Key?

adriangibbons, 2023-05-02:
In `mountain-biking.php` you could replace the API key with your own, or delete/comment out the offending lines of code:
[image]

[image]